clemsonMakerspace/admin #29: Admin Status for all equipment

coledemarco, 2017-02-02:
Cole working with Tim H

ophilli, 2017-02-22:
@coledemarco  Please update as needed.

ophilli, 2017-03-09:
Spoke with Tim H earlier this week - he mentioned there was discussion w/ regards to giving all interns admin access.

It is my recommendation that only one intern be given admin access and that the designated intern be responsible for managing the software installed on the PCs. Having multiple people administering a machine makes it substantially harder to properly secure the machines and make sure malware doesn't inadvertently get installed.

Due to many of our computers being utilized in a very public manner any malicious software that is installed has the opportunity to compromise countless other student's personal info.

coledemarco, 2017-03-09:
Okay if thats what you feel thats fine @ophilli

coledemarco, 2017-03-30:
Mark Helped us giving ophilli admin permissions on all wall monitors. However, we need to establish who has admin on the Makerspace Laptops

ophilli, 2017-04-07:
Mark looked up the laptop ME-MECOMP-L3 in active directory.

It belongs to an OU of Clients/CES.  So CES is the group that manages them.