clemsonMakerspace/admin #49: Figure out Summer/ Fall Staffing Plan

coledemarco, 2017-02-23:


ophilli, 2017-03-09:
Spoke with Dr. Speziale earlier this week and this topic came up in conversation.

It is too late to receive a UPIC sponsorship so Dr. Speziale has recommended hiring an hourly student. She's stated we will need to figure out who might want to work and how many hours we will want them.

It is possible the Watt / CI may be able to help w/ funding.

ophilli, 2017-03-28:
Haleigh reached out to me and expressed interest in the summer internship position. Brad has also mentioned in passing that he is looking for work over the summer.

Nolan-Hoolachan, 2017-03-28:
This should be one of the top priorities of our next meeting.

ophilli, 2017-03-28:
Creative Inquiry summer internship funding typically has an application date of March 30th - we should try to reach out to Dr. Speziale or Cora before then and let them know we want to move forward.

ophilli, 2017-05-13:
Myself and Brad Hord will work in the space May 15th - Aug 4th for a total of 320 hours each.

In the fall the Watt will sponsor the equivalent of 2 FT UPIC interns, and the Makerspace will sponsor 1 FT UPIC equivalent.

This breaks down to 6 part time students. Myself and Cole DeMarco plan to return in the fall meaning we will need to hire an additional 4 students.

closed 2017-05-13 by ophilli