clemsonMakerspace/admin #51: Innovation Forum

coledemarco, 2017-02-23:
-Sponsors
-Faculty (Presenting and Speaking)
-UPIC in the space for people to come by
-----Laser Cut Swag for people that come by

DATE: April 13th

ophilli, 2017-03-02:
4 slots during breaks and lunch
Email from Timh this afternoon

ophilli, 2017-03-15:
Current Availability Wed April 12th
___

#### 11:00 - 11:30am Networking Break
* Nolan
* Cole
* Katie

#### 12:30 - 1:30pm Lunch
* Owen (Leave Early)
* Nolan (Leave Early)
* Cole (1:15 - 1:30)

#### 3:00 - 3:30pm Networking Break
* Owen (Leave Early)
* Cole

#### 5:00 - 6:00 Reception/Exhibits
* Owen
* Nolan (Already Working)
* Katie (Already Working)

ophilli, 2017-03-15:
We will only have two weeks before the event once we return from Spring Break. We need to confirm final promotional designs with the Watt ASAP in order to allow them time to secure materials.

closed 2017-05-13 by ophilli