clemsonMakerspace/admin #53: Vending Machine Facts Sheet

coledemarco, 2017-03-02:
Parts:
-Cromer Facts
-Electronics Prototyping Equip

coledemarco, 2017-03-02:
communicate the actual value this adds to the clemson community

ophilli, 2017-03-09:
Some key facts we need to highlight

1. Non-snack nature
1. Approval from Cromer
1. Cash acceptance will be disabled
1. Products sold will serve students co-curricular pursuits
1. Products will relate to the mission of the Makerspace

#### THE PRODUCTS LISTED BELOW ARE ONLY INDICATIVE OF POSSIBLE OPTIONS FOR STOCKING THE MACHINE AND DO NOT REPRESENT A COMPREHENSIVE OR FINAL PARTS LIST

Example components to support Mechatronics, Wearables, & Internet of Things
* Microcontrollers & Single Board Computers
  * [Arduinos](https://www.adafruit.com/products/659)
  * [Raspberry](https://www.adafruit.com/products/3055) [Pis](https://www.adafruit.com/products/3400)
  * [Particle Photons](https://www.adafruit.com/products/2721)
* Microcontroller & Single Board Computer Add-on 'Shields'
  * [Relays](https://www.adafruit.com/products/3051)
  * [Robotics](https://www.adafruit.com/products/2327)
  * [Environment Sensing](https://www.adafruit.com/products/3194)
* Single Board Computer Accessories
  * [Micro SD for Operating System](https://www.adafruit.com/products/2767)
  * [5v 2.4A Power Supply](https://www.adafruit.com/products/1995)
  * [USB OTG Hub](https://www.adafruit.com/products/2991) w/ [Ethernet](https://www.adafruit.com/products/2992)
  * [Display Cable](https://www.adafruit.com/products/2421)
  * [Keyboard Pack](https://www.adafruit.com/products/1738)
* Sensors
  * [Temperature Sensors](https://www.adafruit.com/products/3251)
  * [RGB Color Sensors](https://www.adafruit.com/products/1334)
  * [Rotary Encoders](https://www.adafruit.com/products/377)
  * [Accelerometers](https://www.adafruit.com/products/2809)
  * [Distance Sensors](https://www.adafruit.com/products/3316)
* Motors
  * [Servo Motors](https://www.adafruit.com/products/169)
  * [Stepper Motors](https://www.adafruit.com/products/324)
  * [Solenoids](https://www.adafruit.com/products/412)
  * [Peristaltic Pumps](https://www.adafruit.com/products/1150)
* LEDs
  * [Wearable Sequins](https://www.adafruit.com/products/3377)
  * [Rings](https://www.adafruit.com/products/1643)
  * [Strip](https://www.adafruit.com/products/1138)
* Miscellany
  * Wire
  * Resistor Packs
  * Capacitors
___
Example components to support CNC machining
  * [Circuit Board Blanks](https://othermachine.co/store/materials/pcb/)
  * [Linoleum Blocks](https://othermachine.co/store/materials/linoleum-blocks/)
  * [Stamp Pads](https://othermachine.co/store/materials/stamp-pads/)
  * [Machining Wax](https://othermachine.co/store/materials/machining-wax/)
  * [Aluminum Bar](https://othermachine.co/store/materials/aluminum-6061-bar/)

#### THE PRODUCTS LISTED ABOVE ARE ONLY INDICATIVE OF POSSIBLE OPTIONS FOR STOCKING THE MACHINE AND DO NOT REPRESENT A COMPREHENSIVE OR FINAL PARTS LIST

ophilli, 2017-03-10:
@kdbryan You should reach out to Dr. Jacob Sorber in Computer Science, as well as some of the professors in charge of Senior Design in Mech E, ECE, and Bio E to help get a diverse faculty response for part recommendations.

closed 2017-05-13 by ophilli