clemsonMakerspace/admin #55: Contact Fumexinc.com

coledemarco, 2017-03-02:


ophilli, 2017-03-09:
I've called Fumex twice and been directed to speak with a Lou but he has been out of office both times.

I've left voicemails but have not received a response yet.

ophilli, 2017-03-09:
Spoke with Kevin from Fumex today.

Plywood is likely causing filters to foul quickly - the vaporized glue from the wood is super hard on them.
- Possible solution is an external 'bag-style' filter that acts before pre-filter.
- Fumex Model FB 1 Pre-filter Assembly
  - $917.00 Stainless steel pre-filter Box
  - $57 Large Capacity bag-filter

Kevin helped me test the pressure ports on the HEPA filter and they appear to be functioning correctly. There are four pressure ports - one on dirty air intake, two below front lid, and one at the base of the box.

Our current pre-filter was quoted at $88, HEPA at $116, and Gas / Odor at $147.

closed 2017-04-04 by ophilli